SRGBmods/QMK-Binaries #2: Redragon K580 bootloader issue

AzatToroyan, 2023-02-18:
Hello. Today I tried installing qmk but after doing so the keyboard stopped typing. Only the red rgb works and the keymap is all different. How can I put it to bootloader to go back to stock fw? What's the combination of bootloader?

naitoshedo, 2023-02-20:
You have to use the flasher or short the MCU pin there is no key codes to go into bootloader mode like most QMK keyboards (Sonix joys).

https://sonixqmk.github.io//SonixDocs/install/#42-entering-bootloader-mode

I would additionally go to the source:
https://www.redragonzone.com/pages/download

AzatToroyan, 2023-02-20:
I have already fixed it with shorting but still the keymap is different and you can't type with the keyboard when the fw is flashed. The VIA one works tho. I just had to go through yarn instalation so didn't bother much

naitoshedo, 2023-02-20:
So to confirm the default bin firmware doesn't work, but VIA does as long as you use their yarn VIA?

AzatToroyan, 2023-02-20:
No, via works without yarn too. I just couldn't control the rgb because I needed srgb plugin which has a doc and in the doc it says to choose my keyboard from the keyboard list but my keyboard wasn't there so I just flashed stock back. I am gonna wait for my keyboard to be added to the list or else I don't know what to do to make it work with srgb

AzatToroyan, 2023-02-20:
I just needed yarn to remap keys

naitoshedo, 2023-02-20:
> No, via works without yarn too. I just couldn't control the rgb because I needed srgb plugin which has a doc and in the doc it says to choose my keyboard from the keyboard list but my keyboard [...]

This is because you have to create the plugin as well: https://docs.signalrgb.com/qmk/building-firmware-from-source#link-to-default-keyboard-plugin

That is what talks to the keyboard, after flashing you can ask for help on creating the plugin after starting work with the Example file. (Adding your VendorId & ProductId to it).

AzatToroyan, 2023-02-20:
I was talking about this doc. Step 9 says to choose my keyboard which I can't because my keyboard just isn't in the list

naitoshedo, 2023-02-20:
Your keyboard doesn't show in Device Information?
https://uploads.developerhub.io/prod/ry3X/y5p211qesstsxoc6rn3p4as28tutopa9sluzm8ksgdxlcs9sife04tbe34hxbgan.png

AzatToroyan, 2023-02-20:
How would it show if I didn't create the plugin? I cant pass through step 9

naitoshedo, 2023-02-20:
Device Information in SignalRGB's settings show all USB connected devices whether they are supported or not by a Plugin.
You are skipping from Step 9 to Link to Default Keyboard Plugin aka Step 17. Everything before this you have technically done as you have the firmware already.

AzatToroyan, 2023-02-20:
Can you help me do it rn? I need to start from flashing and step 1 again

closed 2023-02-20 by naitoshedo