SRGBmods/QMK-Binaries #4: GMMK Pro support

stronk-dev, 2023-03-10:
On the repo I see binaries for `gmmk_pro_rev2`. However, according to their devs that refers to a revision with a different MCU which was actually never put on the market.
I tried flashing this to my GMMK Pro, but as expected the firmware is incompatible (So back to the default QMK firmware for now).
Any way we can get prebuilt binaries for `gmmk_pro_rev1` included in this repo as well?

naitoshedo, 2023-03-10:
Just to let you know the SignalRGB dev who wrote the protocol was shipped a rev2 so..... 

Meanwhile I put up gmmk pro rev1 VIA builds (uid builds as well for it in case the non-uid builds do not work with SignalRGB). UID = Unique Identifier Byte which the dev of the SignalRGB protocol had to add in order to differentiate several Glorious Keyboards due to them reusing USB ProductIDs for multiple keyboard models.

stronk-dev, 2023-03-11:
Hmm I'm having issues modifying the layout using VIA:
```
Vid: 0x320F
Pid: 0x5044
Command name: GET_PROTOCOL_VERSION
Command: 1
Response: 36 27 9 0 20 30 0 20 30 0 20 30 0 20 30 0 20 30 0 20 30 0 20 30 0 20 30 0 20 30 0 0

Name: Error
Message: Received invalid protocol version for GMMK Pro ANSI
Cause: undefined
```
With the [firmware from here](https://www.caniusevia.com/docs/download_firmware) VIA works, but this ofc does not have the SignalRGB patch

stronk-dev, 2023-03-11:
Never mind, this only happens when SignalRGB is running. While SignalRGB is closed the useVIA app works fine, but stops working once SignalRGB takes over RGB controls. Thanks for the help!

closed 2023-03-11 by stronk-dev