SRGBmods/QMK-Binaries #5: Gmmk 2 Flickering and Wrong Color

Lordhengi, 2023-03-16:
Firmware: gmmk2_p96_iso_via.bin
Version: 0.20.1
Software: QMK Toolbox
Issue:
flickering and wrong color of the button 1.

[image]

naitoshedo, 2023-03-16:
Have you tried using the <https://raw.githubusercontent.com/SRGBmods/QMK-Binaries/main/SignalRGB-Plugins/GMMK/GMMK_2_96_ISO_QMK_Keyboard.js> plugin?

Lordhengi, 2023-03-16:
Yes I tried and the problem persists.

naitoshedo, 2023-03-16:
Might want to try the officially supported GMMK2 stuff then: https://docs.signalrgb.com/qmk/supported-keyboards

Lordhengi, 2023-03-16:
I solved the problem by deleting the 120 from the vKeys.
The flickering of button 1 has stopped.

naitoshedo, 2023-03-25:
Has this been resolved or are you waiting on additional help?

closed 2023-04-05 by naitoshedo