SRGBmods/QMK-Binaries #12: XBows Numpad binary

tre4b, 2023-07-26:
Hi there, tried to flash this binary but it does not like it.  Gave me a message about over writing the boot code.   This did not happen with the Knight Plus itself or with the other Knight that I have so it seems to be just this binary.

[image]

Not sure how I would use the suppress code, however I am also not sure that is a good idea as it feels like it could end up bricked.

naitoshedo, 2023-07-30:
I would honestly try using the 0.20.8 VIA build https://github.com/SRGBmods/QMK-Binaries/raw/main/QMK%2BVIA-Firmware/0.20.8/xbows_numpad_via.hex 0.21.x has been a bit problematic for a few users.

tre4b, 2023-08-06:
I've been trying that but QMK just does not seem to like the binary provided.

Atmel DFU device connected (libusb0): Atmel Corp. ATmega32U4 (03EB:2FF4:0000)
Attempting to flash, please don't remove device
> dfu-programmer.exe atmega32u4 erase --force
> Erasing flash...  Success
> Checking memory from 0x0 to 0x6FFF...  Empty.
> dfu-programmer.exe atmega32u4 flash --force "C:\Users\trevor\OneDrive\SystemFiles\Computers\X-Bows Keyboard\Firmware\SignalRGB QMK\Via\xbows_numpad_via.hex"
> Error reading line 1.
> dfu-programmer.exe atmega32u4 reset
Flash complete

I end up with a totally blank Numpad and have to reflash with the stock firmware to get it working again.

tre4b, 2023-08-06:
And this would be because I am a dumb arse and tried to flash them with HTML !     It was a late night last night!!

Have now flashed both perfectly with 20.8 so now for getting the rest of it to work...

tre4b, 2023-08-06:
With these now flashed and a .js file in place the keyboard is not seen.  I am assuming this is because either I did something wrong with the JS file, or that the system is not seeing the file for some reason.  Do you know if there is any way to debug why it is not showing up?

tre4b, 2023-08-06:
Again answering myself - shows up in the logs.   Clearly I have cocked up the JS somewhere.

11:58:55:381 @ [0x8900] - PLUGIN - Failed to load plugin @ C:/Users/trevor/AppData/Local/VortxEngine/app-2.2.32/Signal-x64/Plugins/Razer/Razer_Info_Finder.js - ProductId() returned [0], VendorId() returned [5426]
11:58:58:050 @ [0x8900] - PLUGIN - C:/Users/trevor/OneDrive/Documents/WhirlwindFX/Plugins/XBows Knight Plus SignalRGB Vial.js:56 SyntaxError: Expected token `numeric literal'
11:58:58:050 @ [0x8900] - PLUGIN - Failed to load plugin @ C:/Users/trevor/OneDrive/Documents/WhirlwindFX/Plugins/XBows Knight Plus SignalRGB Vial.js - ProductId() must be of type int, or int[]

Should be able to figure it out from here

tre4b, 2023-08-06:
[image]

Looking good.  Figured out putting the Base65 image in the function too.

naitoshedo, 2023-09-26:
If you would like to provide me the device image I can add it to the QMK Plugins Add-On repo.

naitoshedo, 2023-09-28:
Device plugins & images have been added to the QMK Plugins add-on repo: https://github.com/SRGBmods/qmk-plugins