SRGBmods/QMK-Binaries #19: keychron Q1 pro not recognise

MaximeProulx, 2023-10-19:
After flashing the keyboard with the right bin and putting the right plugin signal RGB still not recognise my keychron q1 pro as a supported device.
I know I am doing the right specs because it's working fine with my keychron k2 pro

naitoshedo, 2023-10-19:
I compiled more recent bluetooth playground firmware and replaced the previous versions, let me know if this helps.

MaximeProulx, 2023-10-19:
Yes I can now see my Q1 pro on signal thanks

closed 2023-10-19 by MaximeProulx

422415, 2024-01-08:
I flashed the 0.14.29 Q1 Pro ANSI firmware and the lighting does work with SignalRGB. However when I try to connect to the keyboard with the VIA windows app or the website usevia.app I get this error:

Fetching v3 definition failed
Device: Keychron Q1 Pro
Vid: 0x3434
Pid: 0x0610

I have made sure to have SignalRGB turned off before trying to connect but still comes up with this error.