SRGBmods/QMK-Binaries #21: Keychron V6 no macros

AnimeFreak0710, 2023-11-08:
Hi, I just flashed my Keychron V6 with knob with the firmware found here: 
https://github.com/SRGBmods/QMK-Binaries/blob/main/QMK%2BVIA-Firmware/0.22.14/K-O/keychron_v6_ansi_encoder_via.bin
After flashing I was no longer able to use macros in https://usevia.app/
It gave me the following error:
> "Your current firmware does not support macros. Install the latest firmware for your device."

[image]

Is this something that was intended as a trade off for enabling the RGB control or is this unintended?
I'm not sure if the slightly more fancy RGB control is worth the trade off for me.
Please let me know if there's anything I can help with to get this fixed if possible, I'd be happy to provide logs and all that.

naitoshedo, 2023-11-08:
Super strange as I do not disable anything on Keychron builds for adding RGB control, I would try the _keychron suffix one instead of the _via one (it is from the newer playground of Keychron aka more recent and yes it has VIA enabled in it as well):
https://github.com/SRGBmods/QMK-Binaries/blob/main/QMK%2BVIA-Firmware/0.22.14/K-O/keychron_v6_ansi_encoder_keychron.bin

AnimeFreak0710, 2023-11-08:
The other firmware (as suggested by @naitoshedo) did **not** help, instead the entire keyboard became unresponsive (no rgb, no keypresses detected, windows complaining about "usb device not recognized". 
Luckily flashing it back to the one in my first comment did seem to fix it, but this also somehow made my keyboard work with macros somehow?? 

I believe something may have gone wrong during my initial flashing of the firmware which fucked it up _just_ enough to make the macros not work.

My apologies for the confusion.

P.S. yes I did try re-flashing the other firmware to see if that wasn't a similar fuckup during the flashing process either, but the issues remained. Should probably be checked or maybe marked as potentially problematic? As this is a different issue than my initial problems it's probably best to make a separate issue to investigate but if you feel differently feel free to reopen this one.

closed 2023-11-08 by AnimeFreak0710

naitoshedo, 2023-11-13:
No other V6 users I work with have encountered this issue, I pushed a complete build of all Keychron's _via & _keychron builds with LED indicator fixes to the 0.22.14 VIA build this evening to make sure all is good and I consider this closed for now.