SRGBmods/QMK-Binaries #22: GMMK 2 96 vol key mapping

jaBEBOP, 2023-12-27:
Hey there, I grabbed the mainline QMK and VIA build for the GMMK 2. The vol up and vol down buttons are the incorrect way around. Also, spacebar + b does not allow me to enter bootloader mode anymore. Can the gmmk 2 builds be updated and are the keys for the bootloader changed now?

https://github.com/SRGBmods/QMK-Binaries/blob/main/QMK%2BVIA-Firmware/0.23.2/0-J/gmmk_gmmk2_p96_ansi_via.bin****

jaBEBOP, 2023-12-27:
Messaged in discord as well

naitoshedo, 2023-12-27:
Firmware corrected: https://github.com/SRGBmods/QMK-Binaries/commit/f6ce2b61a42c51c09311c77b8be6ed0519189074 or you can use VIA to correct the issue without re-flashing the firmware. (Also answered on Discord)

closed 2023-12-27 by naitoshedo