SRGBmods/QMK-Binaries #23: Connection to VIA app gives an error

422415, 2024-01-09:
I flashed the 0.14.29 Q1 Pro ANSI firmware and the lighting does work with SignalRGB. However when I try to connect to the keyboard with the VIA windows app or the website usevia.app I get this error:

Fetching v3 definition failed
Device: Keychron Q1 Pro
Vid: 0x3434
Pid: 0x0610

I have made sure to have SignalRGB turned off before trying to connect but still comes up with this error.

naitoshedo, 2024-01-11:
This is a known error due to all definitions still awaiting [approval](https://www.keychron.com/pages/how-to-use-via-to-pair-with-keychron-q-series-keyboard). I would try the definitions provided there using the Designer tab of VIA. Once VIA gets an update this should be resolved immediately.

422415, 2024-01-12:
I have used the json file supplied by Keychron to load older v2 definitions and it still won't work. If I use the website it seems my keyboard is recognized in the pop up but when I try to connect to it give me this error:
[image]

If I use the windows program it just says "searching for device"

naitoshedo, 2024-01-24:
I haven't forgotten you, it's just that Keychron source hasn't updated at all since this started: https://github.com/Keychron/qmk_firmware/branches

closed 2024-02-01 by naitoshedo