SRGBmods/QMK-Binaries #24: Flashing Keychron K8 Pro leaves keyboard in DFU mode

sudotony, 2024-02-21:
Using QMK Toolbox, when I flash my Keychron K8 Pro with keychron_k8_pro_ansi_rgb_via.bin, my keyboard doesn't restart and seems to remain in DFU Mode. The red LED light under the space bar remains red.

Reverting to stock firmware works fine.

naitoshedo, 2024-02-21:
See if the bluetooth_playground [here](https://github.com/SRGBmods/QMK-Binaries/files/14362620/keychron_k8_pro_ansi_rgb_via.zip) works better than the wireless_playground (current build).

Don't forget to unzip it.

sudotony, 2024-02-22:
Thanks @naitoshedo, flashing wiith the bluetooth_playground gets it out of DFU mode, but the lighting settings from SignalRGB aren't applying. Is that a separate issue I need to chase?

naitoshedo, 2024-02-22:
Is the qmk-plugins add-on in SignalRGB disabled? Perhaps unplug/replug your Keychron K8 Pro for if it's properly flashed and the add-on is enabled it should be picked up immediately.

sudotony, 2024-02-22:
Thanks again for replying and helping out.

The qmk-plugins add-on is enabled and up to date with the latest commit (2 hours ago as of this comment). Unplugging and replugging, disabling and re-enabling doesn't work.

The console read out has this repeating:
[Fri Feb 23 09:18:16 2024] Error - Error: device.write(): HID handle is invalid, confirm that the protocol type selected is HID.
StreamLightingData():354
sendColors():344
Render():93

Perhaps it's a driver issue? I've done the driver installation via QMK Toolbox (it didn't do it automatically), and I've also tried using Zadig to change the driver to WinUSB (v6.1.7600.16385).

I think that's it. I've also got a Q0 numpad that I haven't touched; it registers as HidUsb on Zadig, which seems to align with the error message from SignalRGB. I think I attempted Zadig driver change prior to using the bluetooth_playground.

I'll try to figure out this driver issue and report back.

Also, is there a firmware for the Q0? I can't seem to find it in the QMK-Binaries files.

sudotony, 2024-02-22:
Ok that was easier than I thought. I uninstalled all the various devices (and deleted drivers) that I had installed via Zadig. Unplug, re-plug and now were good. 

Thanks for the help.

(still looking for the Q0 firmware 🤔)

naitoshedo, 2024-02-22:
The q0 (base) and q0 plus can both be found [here](https://github.com/SRGBmods/QMK-Binaries/tree/main/QMK%2BVIA-Firmware/0.23.9/K-O)

sudotony, 2024-02-23:
Thanks for that. 

I tried both [keychron_q0_base_keychron.bin](https://github.com/SRGBmods/QMK-Binaries/blob/main/QMK%2BVIA-Firmware/0.23.9/K-O/keychron_q0_base_keychron.bin) and [keychron_q0_base_via.bin](https://github.com/SRGBmods/QMK-Binaries/blob/main/QMK%2BVIA-Firmware/0.23.9/K-O/keychron_q0_base_via.bin) but both result in a similar 'stuck in dfu' mode after flashing. Reverting to factory firmware fixes.

It's definitely the base model that I have, there are no knob or macro keys.

naitoshedo, 2024-02-23:
See if a default build
[default build](https://github.com/SRGBmods/QMK-Binaries/files/14380680/keychron_q0_base_default.zip) works?

sudotony, 2024-02-23:
Absolute legend, mate. Thank you. Both my K8 Pro and my Q0 are working perfectly with SignalRGB.

closed 2024-02-23 by sudotony