SRGBmods/QMK-Binaries #25: Q1 Pro ISO not recognized in Via / Signal

LRYMND, 2024-02-27:
Hey there,

I just tried flashing the latest firmware to my Q1 Pro Iso but I'm getting the definition error. Loading the json is not working either and SignalRGB lists the device as unsupported. Got the QMK Plugin installed.

If I can provide any additional Info to resolve this let me know!

Cheers

naitoshedo, 2024-03-02:
Don't forget SignalRGB cannot be running when attempting to UseVIA. Even with definition files as they both use the raw_hid_send USB protocol.

LRYMND, 2024-03-02:
I've read about it but via is not even open when attempting to connect to srgb

naitoshedo, 2024-03-04:
If you are capable of doing so hit me up on the [Plugin Discord Server](https://discord.gg/J5dwtcNhqC) to help address this (join #roles channel there and assign yourself the QMK role... then ask in #qmk-tech-support)

closed 2024-03-04 by naitoshedo