SRGBmods/QMK-Binaries #28: Disabling and then enabling QMK on GMMK2 causing flickering leds

efaden, 2024-04-02:
See the video.  Running the latest version.  If I disable within signalRGB it works fine with the hardware modes.  But if I re-enable it I just get flickering.  Changing modes didn't work
.. only fix is to unplug it and plug it black in.

https://github.com/SRGBmods/QMK-Binaries/assets/12599300/0afad85f-03b3-4eb9-95f5-47ace12a5961

efaden, 2024-04-03:
Per discord this is a known bug.

naitoshedo, 2024-04-20:
The latest build 0.24.7 attempts to correct this.

naitoshedo, 2024-04-24:
This is entirely a HID endpoint issue that no matter how many non eeprom writes I work with the easiest solution is either unplugging the keyboard or restarting SignalRGB... The current stable build calls:
`rgb_matrix_mode_noeeprom(RGB_MATRIX_NONE); //Set RGB Matrix to none to allow a re-init.
    rgb_matrix_disable_noeeprom();
    rgb_matrix_enable_noeeprom();
    rgb_matrix_mode_noeeprom(RGB_MATRIX_SIGNALRGB); //Set RGB Matrix to SignalRGB Compatible Mode`
  It's not always flawless but it's the best "adhesive medical strip" (Band-Aid) I can provide right now.

closed 2024-04-24 by naitoshedo