SRGBmods/QMK-Binaries #29: Right arrow key unresponsive to SignalRGB's LED control settings

KINDR3D, 2024-05-31:
Flashed the Keychron q1 pro with both the bluetooth and the wireless bin file, but whatever I try, the right arrow key on my keyboard won't light up. It only turns on when I disable SignalRGB and use the led controller on the keyboard itself (So I know for a fact the LED itself isn't broken)
Any clue what could be the cause of this?

naitoshedo, 2024-05-31:
Is this the ANSI, ISO, or JIS version of Q1 Pro having issues?

KINDR3D, 2024-05-31:
HI, first of all; thanks for your quick reply! I indeed have the Ansi model. It's the version with the knob. I tried restarting restarting SignalRGB but no luck unfortunately. I noticed that it tells me it has 83 controllable leds under Device Information, but the console log tells me it has a Device Total LED count of 82. It also thinks it's the OnePlus version of the keyboard for some reason.

KINDR3D, 2024-05-31:
Nevermind, I didn't install the correct plugin. Everything works now, thank you so much!

naitoshedo, 2024-05-31:
You actually pointed out an issue that the 81 Pro (OnePlus) and the Q1 Pro use the same ProductID so I moved support to the Q1 Pro plugin for both.

closed 2024-05-31 by naitoshedo