SRGBmods/QMK-Binaries #31: Support for Lemokey P1 & P1 Pro

Lyte-vb, 2024-08-28:
The source code is available now. 
https://github.com/Keychron/qmk_firmware/tree/playground/keyboards/lemokey/p1

naitoshedo, 2024-08-31:
Added. [P1](https://github.com/SRGBmods/QMK-Binaries/tree/main/QMK%2BVIA-Firmware/0.14.29-keychron/playground), [P1 Pro](https://github.com/SRGBmods/QMK-Binaries/tree/main/QMK%2BVIA-Firmware/0.14.29-keychron/wireless_playground)

Source of course is available as well: [P1](https://github.com/SRGBmods/KeychronQMK/tree/playground), [P1 Pro](https://github.com/SRGBmods/KeychronQMK/tree/wireless_playground)

closed 2024-08-31 by naitoshedo

MarkDa03, 2025-10-02:
@naitoshedo Hey - HID Configuration not found when we try to use the launcher using the above-mentioned SignalRGB Firmware. Any chance you can create a ticket compatible version one? I need to set my layouts, where i can switch between mac and windows layouts. I cant do that without having access to the launcher. I had to revert back to the original firmware. Also, signalRGB was only able to find the keyboard whilst connected with a cable, not wirelessly. 

Thanks