SRGBmods/QMK-Binaries #35: Support for Keychron Q0 Max

yujinlin0224, 2024-12-11:
Is there any plan to support Keychron Q0 Max?

naitoshedo, 2024-12-11:
The firmware is already there [(keychron_q0_max_encoder_via.bin)](https://github.com/SRGBmods/QMK-Binaries/blob/main/QMK%2BVIA-Firmware/0.14.29-keychron/wireless_playground/keychron_q0_max_encoder_via.bin), a user plugin needs to be created for it directions are linked in the readme for doing this, but for [ease of finding it the link is this](https://docs.signalrgb.com/qmk/srgbmods-qmk-firmware#user-plugin-creation).

naitoshedo, 2024-12-11:
I went ahead and created the user plugin for the Q0 Max now, a restart of SignalRGB will be required for it to auto download the support plugin from the qmk-plugins add-on repo.

closed 2024-12-11 by naitoshedo

yujinlin0224, 2024-12-11:
Thank you very much! I will try it later.

naitoshedo, 2024-12-11:
Keep in mind you do have to use firmware like the one linked above (or compile your own firmware) that supports the SignalRGB Protocol in order for the plugin to work. [Video Tutorial](https://www.youtube.com/watch?v=jU-rYhMrn68)