SRGBmods/QMK-Binaries #36: Keychron Q0 Plus LED Issue

FR1TZZ81, 2024-12-19:
I downloaded and flashed my Q0 Plus with the QMK+VIA-Firmware/0.27.1/keychron_q0_plus_via.bin firmware.  It shows up in SignalRGB, and the LEDs respond to the SignalRGB theme but the entire right hand column LEDs are off.  I flashed it back to the stock Keychron firmware and they work just fine.  Tried it one more time with the keychron_q0_plus_via.bin firmware and the same thing happened(entire right hand column does not light up).  Keys operate as normal, just no LEDs.

naitoshedo, 2024-12-19:
I adjusted the size of the plugin, it was incorrect this should fix the issue.

FR1TZZ81, 2024-12-19:
I thought that might be the case.  I appreciate the quick response.  I will test it this evening and report back.

Thanks!

FR1TZZ81, 2024-12-20:
Looks to be working now.  Thanks again for the quick response and getting it corrected.

closed 2024-12-20 by FR1TZZ81