SRGBmods/QMK-Binaries #37: Support for Monsgeek M1 V5

reallibrom, 2025-01-17:
Is there any chance that someone makes support for the Monsgeek M1 V5?
The existing BIN are for the M1 V3 and don't seem to work at all with the V5.

Here is the original firmware from monsgeek.
https://www.monsgeek.com/faq/how-to-recover-an-m1-v5-us-via-keyboard-after-flashing-m1-v1-qmk-firmware-or-other-wrong-firmware/

Would be very much appreciated. 
Or if there exist another workaround, would be great too.

naitoshedo, 2025-01-17:
Your link is to a binary which does not help with SignalRGB support, SignalRGB QMK support needs a direct mode which we patch QMK source to provide. Monsgeek "sort of" [released source code](https://github.com/MonsGeek/m1_v5) for the M1 v5 except for the wireless source for it so it [cannot compile](https://github.com/MonsGeek/m1_v5/blob/master/uk/m1_v5/post_rules.mk) (line 4).

naitoshedo, 2025-01-17:
The M1_v5 has a lot of changes source code wise that have not been merged with QMK mainline, I attempted to build the firmware for it by forking Monsgeek's wireless repo. They mess with raw_hid_send causing a compilation error when SignalRGB Protocol support is added that I have to work around. I may get around to this at some point but no promises.

naitoshedo, 2025-03-26:
Got the source to build: https://github.com/SRGBmods/QMK-Binaries/issues/40

closed 2025-03-26 by naitoshedo