SRGBmods/QMK-Binaries #38: keychron_q6_max_ansi_encoder_via.bin

peekeesh, 2025-01-26:
Not sure why this doesn't work. If I flash this, it'll say successful but SignalRGB will not work properly. Keyboard is detected in the program but the lighting effects don't work at all.

Now when I try to re-flash the firmware, I would get an error where it states the firmware is corrupt..

> Opening DFU capable USB device...
> Device ID 0483:df11
> Device DFU version 011a
> Claiming USB DFU Interface...
> Setting Alternate Interface #0 ...
> Determining device status...
> DFU state(10) = dfuERROR, status(10) = Device's firmware is corrupt. It cannot return to run-time (non-DFU) operations
> Clearing status
> Determining device status...
> DFU state(2) = dfuIDLE, status(0) = No error condition is present
> DFU mode device DFU version 011a
> Device returned transfer size 2048
> DfuSe interface name: "Internal Flash  "
> Downloading element to address = 0x08000000, size = 99684

naitoshedo, 2025-01-26:
That is a standard error no matter what firmware you flash. Meanwhile I rebuilt the wireless_playground pre-compiled binaries so I would try re-flashing with an updated version (make sure you are downloading the raw file).

peekeesh, 2025-01-27:
Oh ok. I tried to download and flash the updated version but still the rgb lights don't sync with signal rgb. The bin file I downloaded around 1/19 worked though. I should've backed up that file.

naitoshedo, 2025-01-27:
I would highly recommend asking for help on the [plugin discord](https://discord.gg/jY2xrWXXRc).

closed 2025-01-27 by naitoshedo

naitoshedo, 2025-01-29:
The joys of via_command_kb already being used by Keychron in the wireless_playground, expect a beta release to address this, but as of now I reverted back to the 1.0.5 protocol method.