SRGBmods/QMK-Binaries #40: Support for Monsgeek M1 V5 ISO

enlight3d, 2025-03-26:
Hello, may be related to https://github.com/SRGBmods/QMK-Binaries/issues/37

I've seen that you added support for the Monsgeek M1 V5, but I don't find the binary file for the ISO version.

Firmware seems available here : https://www.monsgeek.com/faq/how-to-recover-an-m1-v5-us-via-keyboard-after-flashing-m1-v1-qmk-firmware-or-other-wrong-firmware/

enlight3d, 2025-03-26:
what is "‎QMK+Weirdness-Outdated-Firmware" ??

naitoshedo, 2025-03-26:
Monsgeek did weird things with QMK's source code in the wireless fork of theirs so it gets to live there.

enlight3d, 2025-03-26:
oh alright thanks. Tried to flash it, keyboard is working and recognized.

I used this https://github.com/SRGBmods/qmk-plugins/blob/main/Monsgeek/Monsgeek_M1_V3_Keyboard.js as a base for the SignalRGB plugin.

```js
export function VendorId() { return 0x342d; }
export function ProductId() { return 0xe4c2; }
```

Sadly, no lights. In console I see "Error: Led names and positions vary in length!". That's all.

naitoshedo, 2025-03-27:
Move your plugin out of the User Plugins directory and restart SignalRGB, I added a plugin that should work to qmk-plugins repo now.

enlight3d, 2025-03-27:
should I enroll in preview updates ? I was on signalrgb@2.4.49+b06d84f8.
EDIT: after switching to preview mode go to signalrgb@2.4.53-beta+4fbcf040 and the keyboard was recognized. But still no lights. Tried forcing it to blue (since my current light mode was red) and nothing. No error logs in the console also.

FYI, I'm on discord on the SignalRGB Plugin Test Server if you need faster answers !

naitoshedo, 2025-03-28:
> should I enroll in preview updates ? I was on signalrgb@2.4.49+b06d84f8. EDIT: after switching to preview mode go to [signalrgb@2.4.53-beta](mailto:signalrgb@2.4.53-beta)+4fbcf040 and the keyboard was recognized. But still no lights. Tried forcing it to blue (since my current light mode was red) and nothing. No error logs in the console also.
> 
> FYI, I'm on discord on the SignalRGB Plugin Test Server if you need faster answers !

Feel free to poke me there then, got a couple of folks running it now. Especially Reystar in the #qmk-general channel.

closed 2025-03-28 by naitoshedo

enlight3d, 2025-03-31:
@naitoshedo I flashed again the firmware and it worked !!! Thanks